project-openwrt/openwrt-latest #13: docker在bridge模式下无法访问外网

YYiiEt, 2020-02-28:
设备：树莓派4b，固定lan口ip且关闭dhcp挂在主路由下，其他皆为默认设置。
宿主机可以访问网络，容器在host模式下也可访问网络，但通过bridge却不行。
[image]
[image]
[image]

1715173329, 2020-02-28:
请检索在其他OpenWrt版本下是否拥有相同问题

YYiiEt, 2020-02-29:
在lean的openwrt下有人遇到相同问题，参考：https://www.right.com.cn/forum/thread-1209004-1-1.html

但按照帖子中提供方法
“然后vim /etc/sysctl.conf，添加下列内容：
net.bridge.bridge-nf-call-ip6tables = 1
net.bridge.bridge-nf-call-iptables = 1
net.bridge.bridge-nf-call-arptables = 1
后台修改设置：
Luci>网络>防火墙>转发：接受
Luci>状态>防火墙>重启防火墙
最后ssh执行service dockerd restart”

执行完流程后问题相同
[image]

考虑到问题不普遍，或许是特定的硬件原因吧

closed 2020-02-29 by YYiiEt

1715173329, 2020-02-29:
改了sysctl规则，需要执行 `sysctl -p` 使其生效。

YYiiEt, 2020-03-03:
> 改了sysctl规则，需要执行 `sysctl -p` 使其生效。

感谢解答，不过当时执行过sysctl -p而且重启甚至在编译前添加都无效，已经放弃了