project-openwrt/openwrt-latest #15: ShadowSocksR Plus+无法订阅

ClayMoreBoy, 2020-03-23:
SS/SSR/V2/TROJAN订阅URL无法保存

1715173329, 2020-03-23:
手动点击“保存并应用”即可

ClayMoreBoy, 2020-03-24:
都尝试过不行才发 Issues

ClayMoreBoy, 2020-03-24:
@1715173329

1715173329, 2020-03-24:
上图

ClayMoreBoy, 2020-03-24:
[image]
只有保存按钮 且f12添加`<input class="cbi-button cbi-button-apply" type="button" value="保存并应用" onclick="cbi_submit(this, 'cbi.apply')" size="0">`
[image]
出现
[image]

1715173329, 2020-03-24:
你会用么。。。
[image]
看到旁边的加号了没？点了之后才会添加配置

ClayMoreBoy, 2020-03-24:
加了不能生效

1715173329, 2020-03-24:
ssrplus日志贴上来看看

ClayMoreBoy, 2020-03-24:
[C]: ?
	/usr/share/shadowsocksr/subscribe.lua:23: in main chunk
stack traceback:
/usr/bin/lua: /usr/share/shadowsocksr/subscribe.lua:23: attempt to call method 'get_first' (a nil value)

1715173329, 2020-03-24:
> subscribe.lua:23: attempt to call method 'get_first' (a nil value)

仅在latest上面出现么？

ClayMoreBoy, 2020-03-24:
有人在lede也报过同样的问题
https://github.com/coolsnowwolf/lede/issues/3757
https://github.com/coolsnowwolf/lede/issues/3731

1715173329, 2020-03-24:
这个我估计是你订阅问题，把你的问题和订阅链接通过邮箱发送至 coolsnowwolf@gmail.com ，让大雕看看

ClayMoreBoy, 2020-03-24:
@1715173329 在大雕https://github.com/coolsnowwolf/lede源下编译的固件(SSR+ 173-7)可以订阅
初步解释是否Luci?

1715173329, 2020-03-24:
你可以尝试 coolsnowwolf/lede#3757 的方法修改订阅文件

closed 2020-03-25 by 1715173329